nav-solutions/rinex #23: Output as CRX

larsnaesbye, 2022-09-13:
I'm thinking that being able to Hatanaka-compress output is something we'd like to have, since we can read it.

gwbres, 2022-09-13:
Hello @larsnaesbye,

> I'm thinking that being able to Hatanaka-compress output is something we'd like to have, since we can read it.

it surely is, this requires two major things to be completed

1. [numerical_data_compression](https://github.com/gwbres/rinex/blob/main/rinex/src/hatanaka.rs#L164)
and [text_data_compression](https://github.com/gwbres/rinex/blob/main/rinex/src/hatanaka.rs#L209) must be completed.
This is the easy part and we're almost there. 
The existing test method must probably be improved / unlocked / verified

2. Then you want to actually use it. This takes place in the `to_file()` methods. 
At the moment only the [meteo::to_file()](https://github.com/gwbres/rinex/blob/main/rinex/src/meteo/record.rs#L135) gives decent results if I remember correctly.
This is the [observation::to_file()](https://github.com/gwbres/rinex/blob/main/rinex/src/observation/record.rs#L540) method that you are interested in, it is not completed yet

Eventually, if you want smooth / hidden operations like what we have today when parsing data, you need a `Writer` equivalent of the [existing Reader wrapper](https://github.com/gwbres/rinex/blob/main/rinex/src/reader.rs). The reader wrapper is just there to provide a `.lines()` iteration method, that provides readable text data, even if the source is compressed. So the `Writer` wrapper, would on its counterpart provide a "write line" method and generate compressed data if need be.

I'm not going to be able to work on this topic in the next few weeks, I would say it is reasonnable I can provide something like this by next month, at the expense of the CLI/Data plotting topic

gwbres, 2022-09-16:
@larsnaesbye 

for information, I just pushed the Buffered Writer (hence low level) operations for data efficient production.
I'll try to conclude the hatanaka compression once & for all