nav-solutions/rinex #30: Get `clap` updated

larsnaesbye, 2022-09-16:
There are two CVEs related to the old version of `clap` we include, [as seen in our audit](https://github.com/gwbres/rinex/actions/runs/2992390072/jobs/4798786934#step:3:71).

If we simply update `clap` then tests fail. But that is probably just because we need to change the test parameters.

Moreover, the YAML feature is deprecated and no longer documented, but not removed wholly. So after updating, we should probably look into moving to alternate ways of defining commands. Even though I prefer the YAML approach.

gwbres, 2022-09-16:
Hello @larsnaesbye 

appararently ["argfile"](https://docs.rs/argfile/latest/argfile) can be added on top of newer `clap` revisions, to build a command line from a descriptor.  They don't even provide an example.. Do you happen to know any other option?

larsnaesbye, 2022-09-16:
Well in my own first project that uses `clap` [I went with the newer programmatic approach](https://github.com/larsnaesbye/sbf2rnx-dev/blob/main/src/main.rs#L34). But the YAML feature does work with `clap` v3, it just isn't mentioned anymore.

larsnaesbye, 2022-09-16:
I think I have it figured out now. PR coming up soon...

gwbres, 2022-09-16:
For information, upgrading to v3.2 with "yaml" feature still does work ;)

larsnaesbye, 2022-09-16:
I had problems with calling short names instead of long names when running tests.

gwbres, 2022-09-16:
I am no Rust/Cargo expert, 
but I think if you specify  "clap = 3.2"  (less constrained), it is smart enough to grab the latest "3.2.x" revision. Is it not ?

closed 2022-09-16 by gwbres

larsnaesbye, 2022-09-16:
Yes, and if you specify "3" it will get the latest 3.x.x revision.

```
1.2.3  :=  >=1.2.3, <2.0.0
1.2    :=  >=1.2.0, <2.0.0
1      :=  >=1.0.0, <2.0.0
0.2.3  :=  >=0.2.3, <0.3.0
0.2    :=  >=0.2.0, <0.3.0
0.0.3  :=  >=0.0.3, <0.0.4
0.0    :=  >=0.0.0, <0.1.0
0      :=  >=0.0.0, <1.0.0
```

taken from https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html

larsnaesbye, 2022-09-16:
No, we can easily just specify 3.2 and then only change it when 3.3.x comes out if we would like the new features.

I think I saw they are working on a 0.5 milestone of `chrono`...

gwbres, 2022-09-16:
The help menu is much nicer!

PR #31 (2022-09-16): update clap to 3.2.22
Fixes #30