nav-solutions/rinex #35: strum property `deserialize`probably not used

larsnaesbye, 2022-09-26:
With newer (<0.18) versions of strum, compilation of `rinex` fails. One of the things mentioned in 0.19 is that ["Most of the strum plugins will now error more aggresively on invalid options being used. Historically, the plugins have ignore invalid options, but most of these should error now."](https://github.com/Peternator7/strum/blob/master/CHANGELOG.md#0190).

One of the reported errors is that `deserialize` is unknown:

```bash
error: expected one of: `message`, `detailed_message`, `serialize`, `to_string`, `disabled`, `default`, `ascii_case_insensitive`, `props`
  --> rinex/src/navigation/record.rs:30:32
   |
30 |     #[strum(serialize = "EPH", deserialize = "EPH")]
   |                                ^^^^^^^^^^^
```

Indeed, if I try to rename `deserialize` to anything else under 0.18, it makes no difference. And in newer versions it is not mentioned in the documentation.

The question is if `serialize`enough to handle deserialization as well. The documentation isn't very clear on this...?

gwbres, 2022-09-27:
I agree this is not clear at all.
I just upgraded to 0.24 of both packages, with the [following commit](https://github.com/gwbres/rinex/commit/9e626f14f409789fa1ce5ee6ec5008c84cd35329).
For information, if `deserialize` did not implicitly work, the test `v4_kms300dnk_r_202215910` would fail, as we test there the parsing of several `FrameClass::{IonosphericModel, SystemTimeOffset and Ephemeris}` which can only be interpreted if those key words are recognized

larsnaesbye, 2022-09-27:
Well it seems to work now that `deserialize` has been removed :)

closed 2022-09-27 by larsnaesbye