nav-solutions/rinex #36: Newer versions of `strum` and `strum-macros` work differently

larsnaesbye, 2022-09-26:
When upgrading to newer (>0.18) versions of `strum`/`strum-macros` compilation fails, probably because 0.19+ no longer is silent about errors.

```bash
error[E0599]: no variant or associated item named `from_str` found for enum `Observable` in the current scope
   --> rinex/src/header.rs:466:72
    |
466 |                 if let Ok(observable) = meteo::observable::Observable::from_str(phys_str.trim()) {
    |                                                                        ^^^^^^^^ variant or associated item not found in `Observable`
    |
   ::: rinex/src/meteo/observable.rs:10:1
    |
10  | pub enum Observable {
    | ------------------- variant or associated item `from_str` not found here
    |
    = help: items from traits can only be used if the trait is implemented and in scope
    = note: the following trait defines an item `from_str`, perhaps you need to implement it:
            candidate #1: `FromStr`

```

Is this perhaps because EnumString no longer implements the same function?

gwbres, 2022-09-27:
Hello @larsnaesbye 

I was able to upgrade both packages to version 0.24, with the [following commit](https://github.com/gwbres/rinex/commit/9e626f14f409789fa1ce5ee6ec5008c84cd35329)
let me know if it works on your side

larsnaesbye, 2022-09-27:
Yes, it works now! No problems anymore.

closed 2022-09-27 by larsnaesbye