nav-solutions/rinex #38: YAML support removed in clap 4.0.x

larsnaesbye, 2022-10-04:
I tried with 4.0.9, and the feature has been removed silently. So if we ever need the clap 4 features then we have to move to the alternative ways of specifying commands. Right now, I don't think it's a problem, though.

Some info can be found [here](https://www.reddit.com/r/rust/comments/wagcs7/comment/ii4ccjs/?utm_source=reddit&utm_medium=web2x&context=3).

gwbres, 2022-10-05:
@larsnaesbye 

that is good information. Let's keep this open as a reminder.
I will create an entry in the Todo list to inquire and progress towards a different command line description method.

larsnaesbye, 2022-10-12:
I added the `arg_required_else_help` setting, but now I see that the README changes mention the need to specify `-h`. Do you prefer to throw an error if no args are used?

gwbres, 2022-10-12:
Hello @larsnaesbye 

No, it's just I was not aware that Clap#4 had the feature you provided a couple of weeks ago.
I find it pretty hard to navigate the (huge) differences between all of these versions.
But at last, we're at the latest, and it should remain stable I assume.
It's a shame we can't sort/group command line arguments, I can't seem to find such a feature ?

As you noticed I'm working in the "cli" branch where I updated to Clap4. Let's close this issue once it is merged.
I will only merge when significant progress was made regarding the `-plot` argument, which needs more genericity.

larsnaesbye, 2022-10-12:
> I find it pretty hard to navigate the (huge) differences between all of these versions. But at last, we're at the latest, and it should remain stable I assume. It's a shame we can't sort/group command line arguments together, I can't seem to find such a feature ? this might require an extra library
> 
I will look after a way to group them together. It would be strange to me if they removed that functionality.

gwbres, 2022-10-12:
> I will look after a way to group them together. It would be strange to me if they removed that functionality.

Awesome, thank you very much for your time!
I'm focused on plotting Observations better, unlocking NAV plotting, and concluding file productions

larsnaesbye, 2022-10-12:
I don't see an obvious way of doing it. `ÀrgGroup` does not change the order or display of the parameters.

larsnaesbye, 2022-10-12:
Aha! The best I can do is the [`next_help_heading`](https://docs.rs/clap/latest/clap/builder/struct.Command.html#method.next_help_heading) function, which does indeed make a new section header in the documentation :-)

gwbres, 2022-10-16:
Hello @larsnaesbye

> Aha! The best I can do is the [`next_help_heading`](https://docs.rs/clap/latest/clap/builder/struct.Command.html#method.next_help_heading) function, which does indeed make a new section header in the documentation :-)

:+1:  this looks much much better.

I just merged the cli branch, closing this topic.

Here's what 's new

* (OBS RINEX) plotting improved: cleaned code, introduced dedicated structures & methods
I saw you were inquiring these plotting libs too. I can tell you this has not been easy, and there are still some issues remainig (plot legend for instance). You might benefit at looking at what I do in the Plot Context structure in this application.
The key is to use the `CharState` object, otherwise you keeping facing "scope"  (lifetime) issues, it is impossible to plot inside loops for instance
* Meteo RINEX plotting introduced & validated
* NAV RINEX plotting under development
* it is now possible to process as many files as we want
* it is possible to analyze or visualize (--plot) the results of RINEX preprocessing
* RINEX processing interface has been generally improved, especially the double differentiation

Next week I will introduce benchmarking methods, I will also inquire how to improve some of the Hatanaka methods.
We obviously have too many memory allocations, this will easily be improved, but having a good benchmark is the best thing to determine whether we're going in the right direction or not. I will also try to conclude the NAV and OBS file production method and unlock them in the CLI

see you

closed 2022-10-16 by gwbres

larsnaesbye, 2022-10-16:
Cool that it now allows for processing unlimited number of files!

larsnaesbye, 2022-10-17:
I guess the `cli` and `nav4` branches can be removed now they are no longer ahead of `main`.