nav-solutions/rinex #42: HiFiTime

larsnaesbye, 2022-10-21:
I took a brief look at https://docs.rs/hifitime/latest/hifitime/ and thought that maybe we can use it instead of the `time`and `chrono` crates?

larsnaesbye, 2022-10-22:
Specifically I found out about it through this discussion: https://www.reddit.com/r/rust/comments/y58v4k/hifitime_350_timers_and_chrono_alternative_only/

gwbres, 2022-10-22:
Hello @larsnaesbye 
Indeed, and they're in the process of introducing GNSS timescales, which is exactly what we would need.
I just opened a fork to conclude GNSS timescale definitions, introduce GAL and BDS timescales, and ideally, we would also need conversion methods

larsnaesbye, 2022-10-25:
Looks like it was a welcome addition. 3.6 is out, and so we can begin to replace our 'time' and 'chrono' crates. 
I also really like that HiFiTime has Python bindings. It made me consider working on making bindings for the `rinex` library too, because it could be used by the many people who know Python, but not Rust. We could potentially reach a bigger audience, don't you think?

gwbres, 2022-10-25:
> I'd like to see if I can make Python bindings for the `rinex` crate. It will be a good exercise for me.

awesome, sounds good to me :+1: I will be working on CRINEX compression (rnx2crx) and the NAV plot on my side, by the end of this week, so I will not be able to inquire that topic.

you can push anything you want to this repo. 
I don't mind if you push directly to master, but if you choose this path, try to keep the CI passing. Believe me, there is already a lot of tests and it breaks quite easily :smile:

larsnaesbye, 2022-10-31:
Closing this as we now have HiFiTime in the TODO.

closed 2022-10-31 by larsnaesbye