nav-solutions/rinex #53: `ionex`: looking for 3D ionosphere maps

gwbres, 2022-11-27:
I can't seem to find 3D IONEX maps anywhere (I mean complete maps, not truncated ones). 
Theoretically, 3D IONEX is supported but we don't have any data to prove it does work correctly.

Also, that would be the good basis for several new CLI/Library features, like
- 2D -> 3D upgrading by providing h data
- 3D->2D downgrading by removing h data
- Take altitude into account when visualizing a IONEX
- ... and probably other operations

larsnaesbye, 2022-11-28:
Hmm... We don't have any IONEX 'on the shelves' here at SDFI. But maybe @stenseng can point us in the right direction?

larsnaesbye, 2022-11-29:
Not much luck so far. Wonder if we can create one using [this tool](https://github.com/Hollweg/INPE-TEC-Maps-IONEX)....

gwbres, 2022-11-30:
> Not much luck so far. Wonder if we can create one using [this tool](https://github.com/Hollweg/INPE-TEC-Maps-IONEX)....

I can't seem to find any infos on what the inputs to their script should be

larsnaesbye, 2022-12-02:
Me neither. They mention that the TEC info comes in two data files. But I can find no examples.

larsnaesbye, 2022-12-08:
Hey @gwbres: take a look at the CODE files on http://www.bernese.unibe.ch/download/ , for example: under http://ftp.aiub.unibe.ch/CODE_TEST/IONO/P1/2022/ 

I peeked in one of the files (`COD0OPSRAP_20223070000_01D_01H_ION.INX` after unzipping), and it does mention vertical TEC and has height data (450.0)...

gwbres, 2022-12-09:
>  for example: under http://ftp.aiub.unibe.ch/CODE_TEST/IONO/P1/2022/
> I peeked in one of the files (`COD0OPSRAP_20223070000_01D_01H_ION.INX` after unzipping), and it does mention vertical TEC and has height data (450.0)...

Hello Lars,

these files are like the ones I stored in the repo, where the altitude is fixed.
Maybe I'm misunderstanding, and a RINEX where altitude is not fixed (points given in a volume, not only an area) does not exist. 

Well, actually that is probably not true. If you look at the IONEX specifications, 

```
y/m/d       START OF HEIGHT MAPS
[...]
y/m/d       END OF HEIGHT MAPS
```

Does exist. So the file allows describing a volume, and not only measurement accross an area