nav-solutions/rinex #60: sinex: deprecated calls

larsnaesbye, 2022-11-30:
With chrono 0.4.23 some functions are deprecated: https://github.com/chronotope/chrono/pull/827

```
   Compiling sinex v0.1.0 (/Users/lanch/Desktop/Dev/rinex/sinex)
warning: use of deprecated associated function `chrono::NaiveDate::and_hms`: use `and_hms_opt()` instead
  --> sinex/src/datetime.rs:20:11
   |
20 |     Ok(dt.and_hms(h as u32, m as u32, s as u32))
   |           ^^^^^^^
   |
   = note: `#[warn(deprecated)]` on by default
```

gwbres, 2022-12-05:
We should replace `chrono` by `hifiime` just like we did in the main crate

This crate only uses `chrono::NaiveDatetime`, it gets replaced by `hifitime::Epoch`

Parsing from a string:

```rust
Self {
   [...]
   date: parse_datetime(content.trim())?
}

fn parse_datetime(content: &str) -> Result<chrono::NaiveDatetime [...]
```

becomes something like this

```rust
fn parse_datetime(content: &str) -> Result<Epoch, [...]
     let duration = Duration::from_hours(h)
            + Duration::from_minutes(m)
            + Duration::from_seconds(s)
            + [...] ; 
     Epoch::from_duration(duration, TimeScale::UTC)
```

closed 2022-12-05 by larsnaesbye

gwbres, 2022-12-05:
> Ah yes, replacing with HiFiTime will fix it anyway.

but that still remains to be done 😄