nav-solutions/rinex #66: `ephemeris`: Kepler equations

gwbres, 2022-12-04:
- [ ] introduce Kepler equations, mainly for azimuth / elevation angles determination.
this work is under progress in a dedicated branch
- [ ] exploit that in several library methods
- [ ] exploit that through a `skyplot` view with `rinex-cli`
- [ ] exploit that in enhanced Observation data visualization with `rinex-cli`

gwbres, 2022-12-08:
Hello @ChristopherRabotin,

we can probably achieve something very powerful here, if we have the Navigation/Ephemeris part of this library use the Nyx library internally.

I'm thinking of vehicle attitute determination, movement predictions, filtering, basically all these advanced topics I'm far from familiar with (I'm a _timing_ guy). Like it is done in `Stanford/gnss_lib_py` for example. 

My immediate requirement is to evaluate Elevation and Azimuth angles of identified vehicles, that is what the related PR does. And we will exploit that in new visualizations. But that is probably just a starting point.

As far as Navigation Data is concerned, parsing has been completed months ago, we support up to RINEX4, and the main structure is `src/navigation/ephemeris`. They provide high level methods, for example this one that the PR will introduce:

```rust
let rinex = Rinex::from_file("rinex.txt")
   .unwrap();
let sv_angles = rinex.navigation_sat_angles();
```

ChristopherRabotin, 2022-12-08:
Hi Guillaume!

That would be pretty cool to use Nyx for these computations. I've also been working with two other people to rewrite NAIF SPICE in Rust (with bindings to other languages): https://github.com/anise-toolkit/anise.rs/ . Most of the orbit computations are already there, but the rest of the library is far from complete (current "hopeful" timeline is for it to be usable in Nyx in the next 6 weeks). On my free time, I'm working exclusively on Hifitime and ANISE at the moment, and will then pick up the work on Nyx (switching to hifitime 3.4+ and using ANISE for the input/output of ephemeris data).

In the meantime, for the computation of orbital elements like elevation and azimuth, you can read the math specs of Nyx here: https://nyxspace.com/nyxspace/MathSpec/celestial/orbital_elements/ . In Nyx, you can define "ground stations" that can be used for computing elevation, as is shown here: https://nyxspace.com/nyxspace/showcase/mission_design/orbit_design_ga/#elevation-computation-from-trajectory . 

These could be ways to integrate these computations before I get around to releasing Nyx 2.0.0.

closed 2022-12-11 by gwbres