nav-solutions/rinex #78: Support CNAV ephemeris types

larsnaesbye, 2023-04-14:
### Discussed in https://github.com/gwbres/rinex/discussions/76

<div type='discussions-op-text'>

<sup>Originally posted by **ljbade** March 17, 2023</sup>
With RINEX 4 it is now possible to get broadcast ephemerides for the newer signals such as GPS CNAV or BDS CNAV. However it seems the current implementation only recognises the ephemeris types from RINEX 3 such as LNAV.

I am interested in computing the position difference between the different ephemeris types for the same satellite so it would be useful to add support for:
- GPS CNAV and CNAV-2
- QZSS CNAV and CNAV-2
- BDS CNAV-1, CNAV-2 and CNAV-3

The recent IGS broadcast navigation files have examples of most of these new formats. An example is https://cddis.nasa.gov/archive/gnss/data/daily/2023/brdc/BRD400DLR_S_20230750000_01D_MN.rnx.gz

The IGS file doesn't seem to yet have an example of GPS CNAV-2 yet, which is from the GPS L1C signal.</div>

ljbade, 2023-04-15:
A few more notes based on some of the hacky fixes I made to get CNAV stuff working good enough for what I needed.

I think the JSON file `orbits.json` should be expanded for RINEX 4 to allow an ephemeris type map where the keys are the RINEX names (e.g. "LNAV" "CNAV", "CNV2"). Then as the reader is parsing the file it can use the correct field name map for each type.

I noticed that for some reason there is CNAV formats in the `orbits.json` file but they are marked as version 4, and the legacy ones are marked as version 3. So my hacky fix was just to switch the RINEX version number if I was wanting the CNAV data.

ljbade, 2023-04-15:
Putting this here as a reference, for BDS CNAV this is the field names I ended up using:
```
            {
                "revision": {
                    "major": 4
                },
                "content": {
                    "adot": "f64",
                    "crs": "f64",
                    "deltaN": "f64",
                    "m0": "f64",
                    "cuc": "f64",
                    "e": "f64",
                    "cus": "f64",
                    "sqrta": "f64",
                    "toe": "f64",
                    "cic": "f64",
                    "omega0": "f64",
                    "cis": "f64",
                    "i0": "f64",
                    "crc": "f64",
                    "omega": "f64",
                    "omegaDot": "f64",
                    "idot": "f64",
                    "deltaNDot": "f64",
                    "satType": "f64",
                    "t_op": "f64",
                    "sisaiOe": "f64",
                    "sisaiOcb": "f64",
                    "sisaiOc1": "f64",
                    "sisaiOc2": "f64",
                    "iscB1Cd": "f64",
                    "spare1": "xxxx",
                    "tgdB1Cp": "f64",
                    "tgdB2ap": "f64",
                    "sismai": "f64",
                    "health": "f64",
                    "b1cIntegrityFlags": "f64",
                    "iodc": "f64",
                    "t_tm": "f64",
                    "spare2": "xxxx",
                    "spare3": "xxxx",
                    "iode": "f64"
                }
            }
```

gwbres, 2023-05-17:
>so it would be useful to add support for:  
> GPS CNAV and CNAV-2
> QZSS CNAV and CNAV-2
> BDS CNAV-1, CNAV-2 and CNAV-3

I think we need an extra field to describe the kind of message (Legacy or Civilian NAV)

gwbres, 2023-08-25:
@larsnaesbye, @ljbade,

guys ! I just submitted the foundation for complete modern frames support :rocket:

I'll submit them in separate PRs. This will take quite some time to reach completion, feel free to submit PRs if you want to see this completed sooner.

We should probably add more thorough V4 tests (we only have one file!)

larsnaesbye, 2023-08-25:
Brilliant! What kind of V4 cases do we need in particular?

gwbres, 2023-08-25:
@larsnaesbye,

> Brilliant! What kind of V4 cases do we need in particular?

We're only lacking GPS/QZSS data. 
Our unique V4 file, regarding GPS, only has LNAV frames which means the same content as V2 and V3 frames.  
There are two other types of frames `CNAV` and `CNV2`

closed 2023-08-26 by gwbres

gwbres, 2023-08-26:
RINEX `v0.12.0` embedding this enhancement is being released. Any feedback will be highly appreciated

ljbade, 2023-08-29:
If you are looking for RINEX 4 test data, it seems that the IGS is now starting to publish a lot more RINEX v4 data, we should be able to find some CNAV data I think.

ljbade, 2023-08-29:
Did a quick review and spotted a few mistakes - https://github.com/georust/rinex/issues/146

Apologies for not having some code ready, but I was just doing a visual inspection, I can make a PR a bit later.

hmqucas, 2025-05-04:
Where can the BDS CNAV Beidou ephemeris be downloaded currently?

gwbres, 2025-05-04:
Hello,

(please open dedicated topics)

look online @ CDDIS.gov, if you grab data from recent folders, they have `CNV1`, `CNV2`  and `CNV3` messages.
I can't say for sure, but someone did send me some data recently, they had those frames and he said he downloaded them from that website.

If you find one, you should see the CNAV frames appear, something along those lines:

```rust
use rinex::{
    prelude::Constellation,
    navigation::NavMessageType,
};

let rinex = Rinex::from_file("data.txt").unwrap();

for (key, ephemeris) in rinex.nav_ephemeris_frames_iter() {
   match (key.msgtype, key.sv) {
        (NavMessageType::CNV1, Constellation::BeiDou) => {
              // do something
        },
        (NavMessageType::CNV2, Constellation::BeiDou) => {
              // do something
        },
        (NavMessageType::CNV3, Constellation::BeiDou) => {
              // do something
        },
        _ => {}, // not interested
   }
}
```