nav-solutions/rinex #79: orbits.json GPS CNAV typos

ljbade, 2023-04-15:
In the [`orbits.json`](https://github.com/gwbres/rinex/blob/main/rinex/db/NAV/orbits.json) file there are a few typos I spotted:

Under the GPS RINEX 4 section (actually seems to be for CNAV):
- `deltaNoDot` -> `deltaNDot`
- `uraiNed` -> `uraiEd` (note this is different to the `uraiNed0`/`uraiNed1`/`uraiNed2` fields which are correct
- `iscL1C` -> `iscL2C`
- `iscL1Cd` and `iscL1cCp` should be deleted - don't exist for CNAV, but do for CNV2 (so add back when https://github.com/gwbres/rinex/issues/78 implemented)

larsnaesbye, 2023-05-17:
So, the first three can be replaced without negative effects, I guess? I'll make a PR.

gwbres, 2023-05-17:
> So, the first three can be replaced without negative effects, I guess? I'll make a PR.

Nope, other than the dictionnary fields will eventually get renamed

PR #85 (2023-05-17): Correct RINEX4 section
Fixes #79
commit 772823d1 (2023-05-17): Correct RINEX4 section (#85)

Fixes #79
PR #88 (2023-05-18): change iscL1C to iscL2C
This was omitted previously.

Fixes #79
closed 2023-05-18 by gwbres