nav-solutions/rinex #80: Ephemeris health fields

ljbade, 2023-04-15:
I ended up having to modify the [`orbits.json`](https://github.com/gwbres/rinex/blob/main/rinex/db/NAV/orbits.json) file as I was having problems with the decoding of the health bits. I think this was mainly due to the mix of LNAV and CNAV types in there at the moment. The different ephemeris types tend to have quite different definitions of how they encode the health information.

gwbres, 2023-05-16:
@ljbade 
the orbits descriptor should be more complex, to fully describe modern navigation frames

larsnaesbye, 2023-05-16:
@ljbade Would you like to share the changes in a pull request?

ljbade, 2023-05-17:
Sure I can put up the changes I made, there not pretty but was enough for my immediate need. Shouldn't be too hard to polish it up though.

gwbres, 2023-08-26:
Hello @ljbade,

I started working on this issue [in this branch](https://github.com/georust/rinex/compare/main...nav-health).
It will take a little time for this to conclude, feel free to join the effort.

The documentation are rather unclear on how to correctly interpret bitfields, the only one I found straightforward so far is Galileo. For example, I'm having a hardtime relate the pseudo specifications in the RINEX4.pdf and the QZSS specs.

gwbres, 2023-08-28:
> I agree the RINEX document doesn't always line up clearly with the ICDs. Hopefully I will get some time to check your branch later in the week.

Awesome. Yeah.. it's hard to consider those as real _specifications_: 

[image]

bit[17]..[22] I presume represent the 6 bit mentionned in the QZSS specs, dunno what `w 3 sf 1" means.

I also don't know how to interpret GPS CNAV1/CNAV2, while GPS LNAV is correctly documented

[image]

ljbade, 2023-08-29:
So here is Word 3 of Subframe 1 of the "LNAV" data for QZSS taken from IS-QZSS-PNT-005 section 4.1.2.3 (pg 46):
[image]

Basically they are talking about the "SV HEALTH" 6 bits field.

The next page gives the interpretation of these bits:
[image]

ljbade, 2023-08-29:
I spent some time working on https://github.com/georust/rinex/pull/150/files

The bitflags approach doesn't work well for the ephemeris types were some of the health values are integers rather then single bit flags. Please change this if there is a better way to do it.

I can look at GLONASS and BDS later once you get the basics ready. The details are spread amongst a few different ICDs unforunately. 

ICDs for BDS - http://en.beidou.gov.cn/SYSTEMS/ICD/
on page 2 the Version 2.1 is best one for the D1 and D2 ephemerides, in version 3.0 they removed references to the B2I signal as it is not on the newest satellite generation

ICDs for GLONASS - https://glonass-iac.ru/en/documents/
the website the ICD PDFs were hosted on was blocked from all countries that supported Ukraine in the war early last year but you can get an archive of them at http://web.archive.org/web/20220108173540/https://russianspacesystems.ru/bussines/navigation/glonass/interfeysnyy-kontrolnyy-dokument/

gwbres, 2023-08-31:
@ljbade,

> the CDMA ones cover the CNAV ephemerides

they keep mentionning a possible Glonass `NavMsgType::CDMA` message, supposedly in EOP frames 

[image]

 but it's missing in the main Glonass NAV Table ; is that a typo ? 
 
[image]

 the current `NavMsgType` enum does not have such a field

PR #332 (2025-04-05): NAV flags and other bitfields
## Changes

1. Improves RINEX database description, with special keywords like `flag` to unlock a special parser

2. Makes the RINEX parser (much) smarter and tolerate for null values, when not dealing with non native types
   - native types (mainly, `f64`) apply to Keplerian parameters
   - this RINEX specs apply particularly well to keplerian parameters, where zero values make no sense
   - with the new version, we will store the special flags, even coming from a Zero value. This allows storing `FLAG_OK` indications
    
3. Introduce many subtil structures and e [... cut]
closed 2025-04-05 by gwbres