nav-solutions/rinex #82: Support IGS SP3 files

ljbade, 2023-04-15:
Though SP3 isn't a *NEX file type it is pretty much a companion of the clock RINEX format and used by IGS and others to provide precise satellite orbits.

I can't see any other crate yet that supports SP3 so I think this crate would be a good choice to support it.

The format definition can be found on https://www.igs.org/formats-and-standards/. For some reason the "b" version is missing, can be found at https://files.igs.org/pub/data/format/sp3_glon.txt

Luckily its a fairly simple format type

gwbres, 2023-06-07:
Hello @ljbade,

SP3 parser should be a dedicated crate since it's not a RINEX and it differs a lot. 
Eventually when it becomes available, we should add an `--sp3` argument to the command line tool to make it possible to provide such information. 
Similarly to the `--nav` argument, to combine two RINEX for example

gwbres, 2023-08-28:
Hello @ljbade,

I iniated [this library](https://github.com/gwbres/sp3), that we will merge into the georust community once stabilized.   

It will be easy for the rinex-cli tool, through the `rinex::quality::QcContext` , to take one or several SP3 into account.  
My understanding is, this helps solve PPP. The rust RINEX ecosystem is quite far from that at the moment, so my understanding is, this will not impact before a very long time.

They're might be an interesting aspect if we support real time parsing of all of these IGS datasets, mainly for real time PPP solving, which needs SP3 predictions as soon as they are published.

Feel free to contribute to all of that, lots of topic and I can only achieve so little by myself, especially considering this is no longer part of my main occupation.

ljbade, 2023-08-29:
@gwbres nice, I will take a look.

SP3 is useful when you want precise orbits, so is often used in PPP but can be used for a number of other things. There are even some SP3 files which just contain the "broadcast" orbits translated from the RINEX NAV files, which I found a useful reference when writing my own satellite position calculator.

gwbres, 2023-08-29:
Hello @ljbade,

SP3 is definitely interesting data to have. We'll soon have the ability to load an SP3 pool into the `QcContext` just like in `teqc`.

The main thing it needs now is an interpolatin method, just [like here](https://pypi.org/project/sp3/#interpolation). 
So RINEX pool can compare to the provided SP3 context, and also to combine SP3 together and form a seamless SP3 context.
`rust::nyx-space::Interpolators` might prove handy.

gwbres, 2023-08-31:
The current PR already has the ability to load a pool of SP3 files.  
This allows loading a context spanning several days at once. 

We currently only plot SP3 orbits and clock data, there is no processing involved.  
I will first conclude SP3 interpolation within that crate, once that is done, it will be possible to correlate SP3 data/predictions, to RINEX NAV

closed 2023-08-31 by gwbres