nav-solutions/rinex #146: RINEX v4: Missing GPS CNAV fields

ljbade, 2023-08-29:
I have some time to read through the new `orbits.json` file, and I am comparing it to the modified one I was using from a few months ago and making notes of anything I spot.

GPS CNV2:
- Missing ISCs for L1C:
[image]
- Might need to add two spares here as last two fields are from the next line, for some reason they don't document that

QZSS CNV2:
- Missing ISCs for L1C:
[image]
- Might need to add two spares here as last two fields are from the next line, for some reason they don't document that

BDS CNV1:
- Duplicate keys for `"spare"` - VSCode was highlighting this, should use `spare1`, `spare2`
- Missing field `t_op`: 
[image]
- `b1c` - it might be clearer that this field has special meaning by adding `integrity`, or `if` to the name
- Missing spare between ISC and TGD:
[image]

BDS CNV2:
- Duplicate keys for `"spare"` - VSCode was highlighting this, should use `spare1`, `spare2`
- Missing field `t_op`: 
[image]
- Replace `isc_b1cd` with a spare, then add ISC B2ad between the spare and TGDs:
[image]
- Rename `b1c` to `b2a_b1c`, possibly adding integrity to name like suggestion above:
[image]

BDS CNV3:
- Duplicate keys for `"spare"` - VSCode was highlighting this, should use `spare1`, `spare2`
- Missing field `t_op`: 
[image]
- Delete the `isc_` and `tgd_` values - should go from `sisa1_oc2` to `sismai`:
[image]
- Rename `b1c` to `b2b`, possibly adding integrity to name like suggestion above
- Replace `iodc` with `tgd_b2bi`
- Remove the last two spares and `iode`

To help with the test coverage I have attached the IGS RINEX v4 "BRDC" file I was using earlier in the year. This is a useful file as it has every type of NAV data except for GPS CNV2, and BDS CNV3.
[BRD400DLR_S_20230710000_01D_MN.rnx.gz](https://github.com/georust/rinex/files/12462707/BRD400DLR_S_20230710000_01D_MN.rnx.gz)

I will see if there is the missing ones in any newer files.

ljbade, 2023-08-29:
Looks like latest BRD4 files have BDS CNV3, but still no GPS CNV2:
[BRD400DLR_S_20232400000_01D_MN.rnx.gz](https://github.com/georust/rinex/files/12462761/BRD400DLR_S_20232400000_01D_MN.rnx.gz)

gwbres, 2023-08-29:
That's awesome @ljbade,

that's also the first file we have with `EOP` frames

[image]

gwbres, 2023-08-29:
Ok I'll initiate a draft PR, feel free to join in

I noted several corrections to the JSON descriptor, introduce new test data, etc..

PR #148 (2023-08-29): fix nav-v4 support
- [x] Fix #146 
- [x] Introduce new `test_resource/NAV/V4` file and associated test bench
- [x] Test NAV V4 STO, ION and  EOP logical conditions on all files
- [x] Introduce first EOP tests
closed 2023-08-29 by gwbres