nav-solutions/rinex #147: Test compile fail

ljbade, 2023-08-29:
I was getting an error with `cargo test` on latest main:
```
error[E0432]: unresolved import `rinex::test_toolkit`
 --> rinex/tests/compression.rs:4:9
  |
4 |     use rinex::test_toolkit;
  |         ^^^^^^^^^^^^^^^^^^^ no `test_toolkit` in the root

For more information about this error, try `rustc --explain E0432`
```

I had to add `+#[cfg(feature = "tests")]` to `rinex/tests/compression.rs` and `rinex/tests/production.rs` to get it to run the tests.

Also the tests left two "untracked" git files `rinex/merge.rnx` `rinex/test.crx`. It would be nice to either add these to `.gitignore`, or two write test output files to a temporary directory so not to touch the source tree.

gwbres, 2023-08-29:
@ljbade,

not sure why Problem 1# is happening, I don't have that problem locally, neither does the CI

Problem 2# is correct, both files are produced by merge and production tests.  
Usually I tend to have files deleted right away, we should take care of that

gwbres, 2023-08-29:
@ljbade,

Problem1 shows up, because "cargo test" can no longer run as is, `cargo test --features test` is the default test setup.
That is how the CI test for the crate in "default mode".

The problem here, is how to use shared methos between tests, to build an efficient test environment.

The answer turns out, is to stop using a separate `test/` folder, which does not allow for sharing methods, it can only import the `crate` to be tested.

Having tests embedded within a special library feature is the way to go. That is what I did in `SP3` for example, and I will improve this library similarly

PR #151 (2023-08-30): improve test ecosystem
- [x] fix #147 
- [x] move `rinex/tests` within the crate (`rinex/src/tests` as a module)
- [x] ignore files generated by "doc rs"
- [x] use `std::PathBuf` more often
closed 2023-08-30 by gwbres