nav-solutions/rinex #177: `cargo install rinex-cli` fails

larsnaesbye, 2023-10-12:
Version 0.9.2 doesn't compile properly anymore when trying to install via cargo in debian-slim (building the Dockerfile in https://github.com/larsnaesbye/test-rinextools) . I get 37 errors.

gwbres, 2023-10-12:
hello Lars, 

can you give 0.9.4 a try ? also do you mind copying the errors you are getting, so I can help you out

larsnaesbye, 2023-10-12:
Sure, I get the same 37 errors on macOS X:

```bash
error[E0308]: mismatched types
  --> /Users/lanch/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rinex-cli-0.9.2/src/identification.rs:12:20
   |
12 |     identification(&ctx.primary_data(), pretty, ops.clone());
   |     -------------- ^^^^^^^^^^^^^^^^^^^ expected `&Rinex`, found `&&Rinex`
   |     |
   |     arguments to this function are incorrect
   |
   = note: expected reference `&rinex::Rinex`
              found reference `&&rinex::Rinex`
note: function defined here
  --> /Users/lanch/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rinex-cli-0.9.2/src/identification.rs:18:4
   |
18 | fn identification(rnx: &Rinex, pretty: bool, ops: Vec<&str>) {
   |    ^^^^^^^^^^^^^^ -----------

error[E0308]: mismatched types
  --> /Users/lanch/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rinex-cli-0.9.2/src/identification.rs:14:24
   |
14 |         identification(&nav, pretty, ops.clone());
   |         -------------- ^^^^ expected `&Rinex`, found `&&&Rinex`
   |         |
   |         arguments to this function are incorrect
   |
   = note: expected reference `&rinex::Rinex`
              found reference `&&&rinex::Rinex`
note: function defined here
  --> /Users/lanch/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rinex-cli-0.9.2/src/identification.rs:18:4
   |
18 | fn identification(rnx: &Rinex, pretty: bool, ops: Vec<&str>) {
   |    ^^^^^^^^^^^^^^ -----------

error[E0308]: mismatched types
  --> /Users/lanch/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rinex-cli-0.9.2/src/plot/record/meteo.rs:15:13
   |
14 |         let unit = match observable {
   |                          ---------- this expression has type `&rinex::observable::Observable`
15 |             Observable::Pressure => "hPa",
   |             ^^^^^^^^^^^^^^^^^^^^ expected `Observable`, found a different `Observable`
   |
   = note: `Observable` and `Observable` have similar names, but are actually distinct types
note: `Observable` is defined in crate `rinex`
  --> /Users/lanch/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rinex-0.13.2/src/observable.rs:17:1
   |
17 | pub enum Observable {
   | ^^^^^^^^^^^^^^^^^^^
note: `Observable` is defined in crate `rinex`
  --> /Users/lanch/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rinex-0.14.0/src/observable.rs:17:1
   |
17 | pub enum Observable {
   | ^^^^^^^^^^^^^^^^^^^
   = note: perhaps two different versions of crate `rinex` are being used?

error[E0308]: mismatched types
  --> /Users/lanch/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rinex-cli-0.9.2/src/plot/record/meteo.rs:16:13
   |
14 |         let unit = match observable {
   |                          ---------- this expression has type `&rinex::observable::Observable`
15 |             Observable::Pressure => "hPa",
16 |             Observable::Temperature => "°C",
   |             ^^^^^^^^^^^^^^^^^^^^^^^ expected `Observable`, found a different `Observable`
   |
   = note: `Observable` and `Observable` have similar names, but are actually distinct types
note: `Observable` is defined in crate `rinex`
  --> /Users/lanch/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rinex-0.13.2/src/observable.rs:17:1
   |
17 | pub enum Observable {
[... 632 lines cut ...]
5   + use rinex::split::Split;
    |

error[E0308]: mismatched types
   --> /Users/lanch/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rinex-cli-0.9.2/src/main.rs:497:42
    |
497 |         let html_report = QcReport::html(ctx, qc_opts);
    |                           -------------- ^^^ expected `&QcContext`, found `QcContext`
    |                           |
    |                           arguments to this function are incorrect
    |
note: associated function defined here
   --> /Users/lanch/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rinex-qc-0.1.4/src/lib.rs:119:12
    |
119 |     pub fn html(context: &QcContext, opts: QcOpts) -> String {
    |            ^^^^
help: consider borrowing here
    |
497 |         let html_report = QcReport::html(&ctx, qc_opts);
    |                                          +

Some errors have detailed explanations: E0063, E0308, E0599.
For more information about an error, try `rustc --explain E0063`.
error: could not compile `rinex-cli` (bin "rinex-cli") due to 37 previous errors
```

larsnaesbye, 2023-10-12:
Running it from `main` with `cargo install --path ./rinex-cli` works fine, though.

gwbres, 2023-10-13:
> Running it from `main` with `cargo install --path ./rinex-cli` works fine, though. So that means 0.9.4 fixes the problem.

Indeed we've had Cargo related issues around August / early september, but I also have never seen some of the errors you had right here (&&ref instead of &ref). Glad it works :)

larsnaesbye, 2023-10-13:
It's all good . I will change my set up in the Dockerfile to track the latest version and build instead :)

closed 2023-10-13 by larsnaesbye