nav-solutions/rinex #236: RINEX v2 ionosphere parameters support

fedosgad, 2024-04-24:
In RINEX v2 files header records with labels `ION ALPHA` and `ION BETA` are allowed. They contain coefficients for ionospheric delay calculations for GPS system (GLONASS, which is also allowed in those RINEX versions, does not broadcast any ionosphere-related information and can be safely ignored). Those fields [are checked for](https://github.com/georust/rinex/blob/main/rinex/src/header.rs#L878-L884) by header parser but not handled any further (there are TODO notices).

According to [Navipedia](https://gssc.esa.int/navipedia/index.php?title=Klobuchar_Ionospheric_Model&oldid=11615) and IS-GPS-200D (Figure 20-4), coefficients are used for Klobuchar model calculations. This model is already supported by library. `IONOSPHERIC CORR` field, which replaced `ION ALPHA` and `ION BETA` in RINEX v3, [is handled](https://github.com/georust/rinex/blob/main/rinex/src/header.rs#L884-L911) by header parser which presents it as a separate `IonMessage`. I propose that `ION ALPHA` and `ION BETA` should be handled in a similar manner, adding a `IonMessage` wrapping `KbModel` to resulting `Header`.

I am willing to make a PR for this issue. It would be my first experience with Rust though, possibly requiring thorough code review (I have previous coding experience in other languages, so maybe it's not that bad). Current developers/maintainers, would you like me to make such a PR?

gwbres, 2024-04-25:
@fedosgad,

first off, thank you very much for inquiring and yes any contribution is welcome.
This topic is important because the current state does not allow precise positioning using V < 3 RINEX (so GPS, Glonass and GAL possibly)

fedosgad, 2024-04-25:
Great!
I'll take to it then.

fedosgad, 2024-04-25:
Created #239

closed 2024-04-26 by gwbres

gwbres, 2024-04-26:
Thank you for your contribution, this will be added to V0.16 that I'm publishing this week end