nav-solutions/rinex #266: Panic when Constellation display format

cokkiy, 2024-08-23:
Sample code here:
```
#[test]
    fn constellation_display_test() {
        let collections = vec![
            Constellation::GPS,
            Constellation::Glonass,
            Constellation::BeiDou,
            Constellation::QZSS,
            Constellation::Galileo,
            Constellation::IRNSS,
            Constellation::WAAS,
            Constellation::EGNOS,
            Constellation::MSAS,
            Constellation::GAGAN,
            Constellation::BDSBAS,
            Constellation::KASS,
            Constellation::SDCM,
            Constellation::ASBAS,
            Constellation::SPAN,
            Constellation::SBAS,
            Constellation::AusNZ,
            Constellation::GBAS,
            Constellation::NSAS,
            Constellation::ASAL,
            Constellation::Mixed,
        ];

        for c in collections {
            println!("{}", c);
        }
    }
```
This test always panics on println with msg: "a formatting trait implementation returned an error when the underlying stream did not"

gwbres, 2024-08-24:
Hello @cokkiy,

thank you for picking this up ! the problem was in the `formatting` method, the SBAS (generic/undefined SBAS) was not covered, while there is not reason it should not.

Reproduced, fixed and merged : https://github.com/rtk-rs/gnss/pull/9

I'm upgrading all branches to this. An official release of RINEX is pending, it will be called `0.17.0-alpha`

gwbres, 2024-08-24:
Hello @cokkiy,

thank you for picking this up ! the problem was in the `formatting` method, the SBAS (generic/undefined SBAS) was not covered, while there is no reason it should not.

Reproduced, fixed and merged : https://github.com/rtk-rs/gnss/pull/9

I'm upgrading all branches to this. An official release of RINEX is pending, it will be called `0.17.0-alpha`

closed 2024-08-24 by gwbres