nav-solutions/rinex #270: Cargo build failed.

vicalloy, 2024-09-03:
`cargo build --release --all-features` and failed.
I tried branch `main` and `release-v0.17.0` and get the same error.

I found `gnss-rtk` v0.7.3 removed `Config::static_preset` and I try to set `gnss-rtk = { version = "0.7.2", features = ["serde"] }` and get some new errors.
`error: could not compile nyx-space`

gwbres, 2024-09-03:
Hello @vicalloy, thanks for reporting,

> I found gnss-rtk v0.7.3 removed Config::static_preset and I try to set gnss-rtk = { version = "0.7.2", features = ["serde"] } and get > some new errors.
> error: could not compile nyx-space

I am unable to reproduce, could you post all your steps ? 
I did `git pull origin rinex/main and cargo build --release --all-features"

vicalloy, 2024-09-04:
I think maybe some dependency has released a new version, and your Cargo.lock has been created, so you are using the correct version.

I don't have a Cargo.lock, and I'm getting the wrong version.

```
cargo version
# cargo 1.78.0
git clone https://github.com/georust/rinex.git
cd rinex
cargo build --release --all-features
```

- [build-errors.txt](https://github.com/user-attachments/files/16859481/build-errors.txt)
- [Cargo.lock.txt](https://github.com/user-attachments/files/16859499/Cargo.lock.txt)
    - the `gnss-rtk` version is `0.7.3`
    - I can change version = "0.7.2" to version = "=0.7.2" to force the use of 0.7.2, but I still get the same error.

gwbres, 2024-09-04:
@ChristopherRabotin,

> I don't have a Cargo.lock, and I'm getting the wrong version.

hum.. I'm getting tired of Cargo.. you're right. Even `cargo clean` did not change a thing.
Only after removing the Cargo.lock file, I presume equivalent to `cargo cache` to clear the "cache" did unlock the error.

Now I don't understand how the current problem is able to show up.
The problem comes from a new Anise/Nyx release, which I think upgrades nalgebra.
Nalgebra is one of our core dependencies (to all these crates). 
But RINEX/GNSS depend on `anise = 0.4.2` so it should never try to download `anise = 0.4.3` and attempt the nalgebra update.

gwbres, 2024-09-04:
I think I have a temporary solution, If I set :

`nalgebra = { version = "=0.33.3"} `  
`anise = { version = "=0.4.2"} `  

to make sure we remain in both versions, it seems to work fine on my side

gwbres, 2024-09-04:
I'm setting up a daily workflow to detect errors "sooner"

vicalloy, 2024-09-04:
Thanks, it worked.
Note:
`nalgebra` should be `=0.33.0`, `0.33.3` does not exist.

gwbres, 2024-09-04:
Let's try the new #272 pending versions (`-alpha-1` on all impacted libraries / applications)

gwbres, 2024-09-04:
@vicalloy,

let me know if we can confirm this is fixed.
Now that we have a daily workflow, we will hopefuly detect similar "problems" earlier

vicalloy, 2024-09-05:
> @vicalloy,
> 
> let me know if we can confirm this is fixed. Now that we have a daily workflow, we will hopefuly detect similar "problems" earlier

I tried the main branch, and it worked.

gwbres, 2024-09-05:
Thank you all ! 
Similar errors will be detected sooner thanks to the daily job now

closed 2024-09-05 by gwbres