nav-solutions/rinex #285: The tracking schedule and trk_midpoint are not correct in the CGGTTS Conventional Observation schedule.

vicalloy, 2024-12-24:
According to [CGGTTS-Version 2E](https://www.bipm.org/documents/20126/52718503/G1-2015.pdf/f49995a3-970b-a6a5-9124-cc0568f85450)

1. `A minimum of 3 minutes was then foreseen in the BIPM schedule to give to the receiver the time to track the following scheduled satellite.`   
2. `The data to be used are the pseudorange measurements collected during the 13 minutes starting with the date specified in the BIPM tracking schedule.`

If the Common-View Duration is 960 seconds, the `trk_duration` should be 780 seconds (13 minutes * 60 seconds). The `trk_midpoint` should be calculated as `next_release - 3 minutes - (780 seconds / 2)`.

https://github.com/georust/rinex/blob/fc20d8c178e08a235a7430fc5a49d9c204a5b3bd/rinex-cli/src/positioning/cggtts/mod.rs#L422

vicalloy, 2024-12-27:
Thank you for your work, but changing only the `trk_midpoint` is not sufficient.  
The CV period is 16 minutes, which includes 3 minutes of preparation time and 13 minutes of tracking time. We should skip the 3-minute preparation time.  The `TRKL` in CGGTTS should be `780`, not `960`.

I tried to create a pull request(https://github.com/georust/rinex/pull/288) to fix this issue.  
I tested it with my data, and it works. However, some variable names may still seem odd, which is why I didn’t fix  them initially.
- `cggtts::track::Scheduler`:  
    - `trk_duration` should be renamed to `cv_duration`.  (should keep `trk_duration`?)
    - `bipm_tracking_duration` should be renamed to `bipm_cv_duration`.  
- `cv_duration` in CGGTTS is `TRKL`.

```
                  tracking start point                           
                    │                                            
      prepare(3min) │                                            
                    │  tracking(13mins)         │                
      │         │◄──┘                           │                
 ─────┼─────────┴───────────────────────────────┼────────        
      │                                         │                
             cv period(16min）                                    
      ▲                                         ▲                
      │                                         │                
    cv start point                       cv/tracking end point   
```

gwbres, 2025-02-09:
Hello @vicalloy, to keep you updated:

* the upgraded rinex backend is merged into the master branch, in the `rinex-backend` branch.
* In this branch, the applications are upgraded to the new internal library
*  the PPP is updated and it works fine
*  the CGGTTS mode is updated and being simplified / improved
   - the logs now format Epochs with `{:?}` which displays everything in UTC timescale, which is compatible with CGGTTTS
   - track collection and PVT solving are two different things. Even during dead time (no cggtts collection) we keep the PVT solver running to preserve the filter state
   - the CGGTTS reference signal logic is being corrected (C1 for CPP navigation technique, L1 for PPP navigation technique)
   - the period, collection period, warmup period and track mid point are being corrected/simplified

gwbres, 2025-02-19:
The applications (like `rinex-cli`) are moved to the `rtk-rs` organization, this repo will continue hosting the parser and low level libraries. This is already in place and there is no discontinuity in the service. Refer to the splitted repositories and documentation (being updated) from now on

closed 2025-02-19 by gwbres