nav-solutions/rinex #286: If the JPL database cannot be accessed, the creation of QcContext will fail

vicalloy, 2024-12-24:
https://github.com/georust/rinex/commit/d0602bb07084664d628d0c08631dedd3ef6f7dac#diff-67600567e036bbcbf6e0f241a23a11f60131adb0d7f1d30f3e2e9acd75b8efbd  After this commit, creating CGGTTS will fail if there is no network access. 

If the JPL database cannot be accessed, meta_almanac.process(true)?; will raise an error, causing the creation of QcContext to fail.

gwbres, 2024-12-24:
Hello,

thank you for helping improving our tools once again;

that's exactly what we're trying to avoid actually.. we should always be able to rely on an offline model.
I presume this scenario has not been tested yet, I'll give it a try tomorrow

gwbres, 2025-01-09:
@vicalloy, I will try to provide an answer to this next week end

gwbres, 2025-01-10:
I'm providing a solution that works well.
Please refer to the commit logs.
This will be documented in the next released version

PR #290 (2025-01-10): Jpl gh 286
Fix offline/online deployment options
- [x] fixes #286 
- a new --jpl-bpc option is introduced in RINEX-Cli
  Which always requires internet access and IS THE ONLY OPTION
  to keep the JPL model up to date.
- the local storage is preserved and allows fast deployments
  (after a first successful deployment)
- for example, in the past few days the JPL servers have been down.
  
When deploying without --jpl-bpc:
  - the session will rely on the previous JPL model if it was already downloaded.
    If this model is too old and servers remain down, you should delete it !
   - the session  [... cut]
closed 2025-01-10 by gwbres