nav-solutions/rinex #356: CRINEX to RINEX panics on Phase Lock loss

gwbres, 2025-05-25:
## Issue report

The CRINEX to RINEX decompressor panics on eventful Epochs, for example when the PLL lock is lost and that kind of event is being described

## Example

```C
rinex-cli --fp CRNX/V3/HKWS00HKG_R_20250300000_01D_30S_MO.crx.gz

byte index 40 is out of bounds of `                                 4213`
stack backtrace:
  4: rinex::hatanaka::decompressor::DecompressorExpert<_>::next_sv
```

MrCry0, 2026-10-06:
Hi @gwbres and @Artoria2e5, I looked at this on current main and reproduced the exact panic: `next_satellite()` slices the epoch descriptor from the position of the first satellite (byte 40 in RINEX 3), and panics when the recovered descriptor is shorter than that. A bounds check fixes the panic, but I do not know why the descriptor ends up that short, and without that I would only be hiding a decompressor state desync behind an error.

HKWS00HKG_R_20250300000_01D_30S_MO.crx.gz is not in nav-solutions/data. Could one of you provide one of the following?

1. The file itself, or a link to where it can be downloaded.
2. If that is not possible, an excerpt of the CRINEX around the failure: the file header, the last intact epoch (its `>` line, the clock line and the data lines), and the following 2 or 3 epochs up to and including the line that triggers the panic.
3. The matching CRX2RNX output for the same epochs, so the expected result is known.

A few lines are enough as long as the epoch descriptor lines are complete and unmodified. With that I can add a regression test and find out whether the real problem is upstream of the slicing.

gwbres, 2026-10-06:
@MrCry0,

i found that very file in my archives, here it is. 
I will be available tonight to further investigate, I'm busy at work at the moment.
Note that the CRX2RNX tool has been released and it comes with all your patches

[HKWS00HKG_R_20250300000_01D_30S_MO.crx.gz](https://github.com/user-attachments/files/33109044/HKWS00HKG_R_20250300000_01D_30S_MO.crx.gz)

MrCry0, 2026-10-07:
@gwbres, thank you for finding and sharing the file.

I ran it against current main (d6b5642d, which includes #453) and it still panics, so this is a different problem from the one #453 fixed. The panic is in the epoch descriptor parsing (src/hatanaka/decompressor/mod.rs, `next_satellite`: byte index 40 out of bounds of a 34 character string).

The cause is a special event epoch. Line 140865 of the decompressed text is `>` followed by `4213`, that is event flag 4 with 213 header lines following (the "# OF SATELLITES" / "PRN / # OF OBS" block at the end of the file). The decompressor has no handling of event epochs: it takes the leading `>` for a full epoch re-initialization and then tries to read satellites from a 34 character descriptor.

I have a fix for it in the decompressor: an event epoch is now published without satellites and its announced records are forwarded unchanged. With it your file decompresses completely (2880 epochs). I will open a PR shortly.