nav-solutions/rinex #364: RINEX to UBX-MGA converter

jheaff1, 2025-06-14:
Hi,

Awesome library!

Given the recent announcement that the u-blox AssistNow service will be terminated in May 2028 (for M8 devices), it would be great to have a tool which can converter ephemeris and almanac RINEX to UBX-MGA messages, as a replacement for the AssistNow service.

gwbres, 2025-06-14:
Hello @jheaff1,

interesting. I think in the long term, the combination of the `ublox` parser and this framework should enable pretty much everything. Including assistance / correction upload, and it's going to be super interesting.

I just initiated a branch to play with this topic, it's named `ublox`, it creates a new crate option to obtain a `RNX2UBX` streamer.
As similarly done for `BINEX`.  I will keep this branch alive until we reach some kind of conclusion

Before going any further, please note that:

- the ublox parser is currently limited in its MGA support, there's still a bit of work if I recall correctly
- MGA is deprecated and was developped for GPS Only. SFRBX is the new approach and it covers Multi-GNSS.
- SFRBX is also work in progress, it's correctly supported for GPS 

Currently I am not able to achieve anything that you want, because _apparently_ the `ublox` library does not offer MGA / SFRBX forging, only parsing. It seems that _forging_ is done using the `XXXBuilder` structure, and it is not provided for those frames. I'm not sure what needs to be done to unlock this. Join us on this channel or ublox discord channel

PS : in case you're interested, we also have [UBX2RINEX](https://github.com/rtk-rs/ubx2rinex), the mirror operation, as a command line application

jheaff1, 2025-06-16:
Hi @gwbres ,

Thank you for the prompt response. Great to hear you've already started to play around with implementing a Rinex to UBX streamer.

It seems to me that MGA messages do support constellations other than GPS, ie see `UBX-MGA-GLO`, `UBX-MGA-GAL` and `UBX-MGA-BS` - https://cdn.sparkfun.com/assets/learn_tutorials/8/6/9/u-blox8-M8_ReceiverDescrProtSpec__UBX-13003221__Public.pdf#page269.

At any rate, looking forward to seeing how this progresses

gwbres, 2025-06-17:
Ok so first off, the synthesis of UBX-MGA/SFRBX was  provided by my last contribution and this has not been released yet. So we should reindex the branch to the following, this should enable UBX-MGA synthesis. I don't have time to try this as of right now. I don't know if a ublox release will happen in the next few days/weeks, I will try to push for it

```toml
ublox = { rev = "2fa6761b2ccd4344326ad55f0dadc5965e06e957" }
```

or 

```toml
ublox = { branch = "master" }
```

> It seems to me that MGA messages do support constellations other than GPS, ie see UBX-MGA-GLO, UBX-MGA-GAL and UBX-MGA-BS 

Yeah I know, I provided those frames to the library years ago, but it's only recently that I have some little time to look into this, and also got to a stage where we can think of using them. So, to be more accurate, I should say the support exist but principle remeains to be demonstrated.

Let's reindex the branch, see if we can take advantage of the frame forger. The interface we provide should provide both `MGA` (GPS/GLO deprecated and limited frames) and SFRBX (modern) frames. We can start with MGA only to make it simple

gwbres, 2025-07-09:
Will get back to this next week hopefully, keep you updated

jheaff1, 2025-07-10:
Thanks for keeping me updated, it’s very much appreciated! Looking forward to seeing what you implement

gwbres, 2025-09-08:
The ublox library does not allow frame forging, it only allows frame decoding.
It's due to current design choices and the "desire" to use macros everywhere to simplify packet definitions.
The macros were built for frame decoding, not forging.

Currently, I will be able to provide a new RINEX option, to encode RINEX to GNSS protos (GPS, QZSS, ...) but not UBX frames

gwbres, 2025-09-08:
OK this is starting to work !
refer to [these lines](https://github.com/nav-solutions/rinex/blob/ublox/src/tests/ublox.rs#L1) for a compelling example, until further documentation.

I have to point to a specific branch of the UBLOX lib [untill the library is corrected](https://github.com/ublox-rs/ublox/pull/173/commits/843f7fc69fda64896b74689d1fb2473b035ea111), this will prohibit a release until then.

Stay tuned I hope to add MGA-IONO and other helper frames.
I'm also working on GPS/QZSS binary encoding, once this is validated, we can encode Ephemeris frames as raw GPS/QZSS bytes,

Could you elaborate on what we could do with those ?
I presume we can then use a RINEX file to quick start a hotstarted receiver ?
But you can also think of multi receiver network, where you could have receivers "help" one another by exchanging MGA by means of intermediate RINEX

jheaff1, 2025-09-09:
Fantastic work, thank you! 

My use case is to download ephemeris/almanac from the constellation provider’s website (eg NASA for GPS), which I believe are in RINEX form. Then using your code I would translate the RINEX files into ublox messages and send these to a ublox receiver to reduce the time to first fix.